sscs-ose/sscs-chipathon-2025 #24: run_GL.sh fails to install packages on ARM64

MrHighVoltage, 2025-06-19:
@amisapta15 

Again, we unfortunately run into an issue. While building for ARM64, we encountered the issue that netgen in conda is only available for x86_64.

As a hot-fix, we decided to conditionally only install the venv on x86_64 and skip it on arm64.

MrHighVoltage, 2025-06-19:
And I don't know how strict it is with dependencies, because we also have magic and ngspice in the container already installed. Would make the additional installs even a bit more compact....

amisapta15, 2025-06-19:
Yep. Its the same. Most of the Glayout tools are already installed. In past, we faced some dependency issues (mainly due to gdsfactory) and decided to go with a particular version choices. But as far as Netgen and Ngspice is concerned, should be fine.

MrHighVoltage, 2025-06-19:
Okay, we just tested and it still pulls netgen etc. as a dependency. For now (so we have an image for tomorrow) we will not install anything during the build process, as it can anyways be installed when running run_GL.sh.

amisapta15, 2025-06-19:
Sure! Thanks

MrHighVoltage, 2025-06-20:
I just checked, this is what it looks like for me, run in a freshly installed conda env:

[image]

Of course, netgen is now also just to find in PATH, so this check has to be changed...

MrHighVoltage, 2025-06-20:
Also, if you didn't, pull the container again. We pushed a new version yesterday evening.

amisapta15, 2025-06-20:
oh! Thanks. I will pull again. This is with the "chipathon_install.sh" script, right? because this seems to find magic and netgen in the `headless/conda-env`, which is not installed with `run_GL.sh`

amisapta15, 2025-06-21:
@MrHighVoltage Which file is sourced when I start a terminal in Docker? i.e. when I start a terminal with Docker and do `whereis magic` it points to `foss/tools/` (alongside /etc/ for some) but I couldn't seem to get this in the interactive terminal of the Jupyter server. 

```bash
$ whereis magic 
magic: /etc/magic
$ whereis netgen
netgen:
$ whereis klayout
klayout: /usr/local/klayout
```
and 

```python
cmd = f'bash -c "magic --version"'

subprocess.Popen(
    cmd, 
    shell=True, 
    stdout=subprocess.PIPE, 
    stderr=subprocess.PIPE
)
```
doesn't even return anything. 

These we don't want to use, right?. The best current solution that I am thinking of now is, to pre-append `/foss/tools` in $PATH on top of the example tutorials and examples. I couldn't seem to find any other general solutions.

amisapta15, 2025-06-23:
@MrHighVoltage any suggestions? particularly for the permission denied issue?

amisapta15, 2025-06-23:
@MrHighVoltage Can you please try the same with `start_chipathon_vnc.sh` and then running `run_GL.sh`? Then, try the commands above in the Jupyter terminal? Just to see if you get the same error as mine with permission denied?

MrHighVoltage, 2025-06-23:
Also, the directories you add to PATH are not searched recursive. In /foss/tools there are no binaries, so adding this does nothing.
But, there is /foss/tools/bin, which holds symlinks to all relevant binaries, so this is the path you are probably looking for.

amisapta15, 2025-06-23:
@MrHighVoltage, I am unable to replicate your steps, which is very strange. I am getting stuck at "running run_GL.sh" from a terminal in the VNC session itself, not sure why? 

Ok. My Steps are as follows: (1) I use the `start_chipathon_vnc.sh` to build the repo, and can access the desktop env through a browser. There, I open a terminal and I run `bash /dockerstartup/scripts/run_GL.sh`, which fails 
not sure where I am deviating from your run. Please see attached
[image]

```
/foss/designs > ./dockerstartup/scripts/run_GL.sh
bash: ./dockerstartup/scripts/run_GL.sh: No such file or directory
/foss/designs > sh /dockerstartup/scripts/run_GL.sh
Existing GLdev environment detected. Skipping setup.
Launching Jupyter Lab...
/dockerstartup/scripts/run_GL.sh: 53: source: not found
/foss/designs > ./ /dockerstartup/scripts/run_GL.sh
bash: ./: Is a directory
/foss/designs > bash /dockerstartup/scripts/run_GL.sh
Existing GLdev environment detected. Skipping setup.
Launching Jupyter Lab...
Traceback (most recent call last):
  File "/headless/conda-env/miniconda3/bin/conda", line 12, in <module>
    from conda.cli import main
  File "/headless/conda-env/miniconda3/lib/python3.13/site-packages/conda/__init__.py", line 7, in <module>
    from json import JSONEncoder
  File "/usr/lib/python3.12/json/__init__.py", line 106, in <module>
    from .decoder import JSONDecoder, JSONDecodeError
  File "/usr/lib/python3.12/json/decoder.py", line 3, in <module>
    import re
  File "/usr/lib/python3.12/re/__init__.py", line 125, in <module>
    from . import _compiler, _parser
  File "/usr/lib/python3.12/re/_compiler.py", line 18, in <module>
    assert _sre.MAGIC == MAGIC, "SRE module mismatch"
AssertionError: SRE module mismatch
Error in sys.excepthook:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/apport_python_hook.py", line 228, in partial_apport_excepthook
    return apport_excepthook(binary, exc_type, exc_obj, exc_tb)
  File "/usr/lib/python3/dist-packages/apport_python_hook.py", line 56, in apport_excepthook
    if not enabled():
  File "/usr/lib/python3/dist-packages/apport_python_hook.py", line 24, in enabled
    import re
  File "/usr/lib/python3.12/re/__init__.py", line 125, in <module>
    from . import _compiler, _parser
  File "/usr/lib/python3.12/re/_compiler.py", line 18, in <module>
    assert _sre.MAGIC == MAGIC, "SRE module mismatch"
AssertionError: SRE module mismatch

Original exception was:
Traceback (most recent call last):
  File "/headless/conda-env/miniconda3/bin/conda", line 12, in <module>
```

amisapta15, 2025-06-23:
This (`bash /dockerstartup/scripts/run_GL.sh`) runs as usual from the terminal of Docker from the Docker Desktop Exec terminal.  Then, from the Jupyter server -> Terminal, this works 

```
os.environ["PATH"] = os.environ["TOOLS"]+'/bin:'+ os.environ["PATH"]

cmd = f'bash -c "magic --version"'
            
proc=subprocess.Popen(
    cmd, 
    shell=True,
    text=True,
    stdout=subprocess.PIPE, 
    stderr=subprocess.PIPE
)

print(proc.stdout.read())
```

This makes sense as I am running the `run_GL.sh` from the Exec tab of the Docker-Desktop, so the `/headless/.bashrc` never sourced in my pipeline thus I was unable to find the path until I added `/foss/tools/bin` in the $PATH

Do you then see any better solution?  Or, I just ass the path and proceed with rest of my stuff as usual?

amisapta15, 2025-06-25:
@MrHighVoltage I found a way more robust than the previous! Adding the following python code add all the things that you source from `.bashrc`

```python
import os
import subprocess

# Run a shell, source .bashrc, then printenv
cmd = 'bash -c "source ~/.bashrc && printenv"'
result = subprocess.run(cmd, shell=True, text=True, capture_output=True)
env_vars = {}
for line in result.stdout.splitlines():
    if '=' in line:
        key, value = line.split('=', 1)
        env_vars[key] = value

# Now, update os.environ with these
os.environ.update(env_vars)
```

amisapta15, 2025-07-04:
The primary working solution now is to build these tools from source.  Either through Python or in bash

```python
import shutil, os
# Check if the executable exists in $PATH
if shutil.which("magic"):
    print("magic is available in $PATH.")
else:
    # Install magic Layout Editor
    !git clone https://github.com/RTimothyEdwards/magic.git
    %cd /content/magic
    !source ./configure
    #os.system("./configure")
    !make
    !make install
    %cd ../
```

or 
```bash
git clone http://git.code.sf.net/p/ngspice/ngspice && \
    cd ngspice && \
    ./compile_linux.sh
```

This issue can be closed now.

MrHighVoltage, 2025-07-04:
This is not necessary. magic is definitely installed in the container (see `/foss/tools/magic`, the binary is in `/foss/tools/magic/bin`) and it just needs to be in the correct path to find the binary.
Also, this version is rather recent, usually at the time of release of the container it is just about the latest commit.

amisapta15, 2025-07-04:
Absolutely agree, and the relevant Python code for updating the environment variable is provided in the previous comment. I guess we can close this issue now. 

> [@MrHighVoltage](https://github.com/MrHighVoltage) I found a way more robust than the previous! Adding the following python code add all the things that you source from `.bashrc`
> 
> import os
> import subprocess
> 
> # Run a shell, source .bashrc, then printenv
> cmd = 'bash -c "source ~/.bashrc && printenv"'
> result = subprocess.run(cmd, shell=True, text=True, capture_output=True)
> env_vars = {}
> for line in result.stdout.splitlines():
>     if '=' in line:
>         key, value = line.split('=', 1)
>         env_vars[key] = value
> 
> # Now, update os.environ with these
> os.environ.update(env_vars)