sscs-ose/sscs-chipathon-2025 #35: Issues with Opamp Design Challenges 1, Challenge.md file

shubham-analog, 2025-06-28:
1. The Google Doc link [**📊3rd Weekly**](https://docs.google.com/presentation/d/11iUuCnZQZhC-76pMEIqsWOjoYRWxP6tA/edit?slide=id.g369dd00d293_0_563#slide=id.g369dd00d293_0_563) has access issues, please correct it.
2. Supply Specification: Supply is mentioned as 1.8V. We are using 3.3V.
3. Power Specification: Is it associated with 1.8V supply ? If yes please update for 3.3V supply (will allow ~1.83x more current). 

@amisapta15 @msaligane @AL-255

shubham-analog, 2025-06-28:
Also I don't know if you want to increase the output swing specification (considering it was set for 1.8V) ?

amisapta15, 2025-06-28:
> Also I don't know if you want to increase the output swing specification (considering it was set for 1.8V) ?

@AL-255

AL-255, 2025-06-28:
1. It should be the same link as the one in https://github.com/sscs-ose/sscs-chipathon-2025/tree/main/schedule. Currently waiting for the committee to release the slides.
2. The typical core voltage for 180nm is 1.8V, which is where the specification originates. There's no tight restriction on linearity, so using 1.8V will actually make the design easier.  
3. I don't understand the question. The specification is on Power, so using 3.3V will further limit your current (1.83x less).

shubham-analog, 2025-06-28:
> 3. I don't understand the question. The specification is on Power, so using 3.3V will further limit your current (1.83x less).
I am assuming that power spec is defined taking 3.3V supply into account. 
We are using gf180mcuD flavor which does not any core devices. We only have access to 3.3V fets.

AL-255, 2025-06-28:
You can use 3.3V FETs for 1.8V designs too