sscs-ose/sscs-chipathon-2025 #42: start_chipathon_x.sh : unable to open display

radityankn, 2025-07-08:
So from the #39 yesterday, I decided also to try the x11 version of the script. The script appeared to have no errors coming from outside the container. But it failed to run, and here's what I found when running `docker container logs iic-osic-tools_chipathon_xserver_uid_1000` : 

```  
[INFO] USER_ID: 1000, GROUP_ID: 1000
[INFO] Final PATH variable: /foss/tools/bin:/foss/tools/sak:/usr/local/sbin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/foss/ tools/klayout:/foss/tools/libman:/foss/tools/osic-multitool:/foss/tools/rftoolkit/bin:/foss/tools/yosys/bin
[INFO] Final PYTHONPATH variable: /usr/lib/python312.zip:/usr/lib/python3.12:/usr/lib/python3.12/lib-dynload:/usr/local/lib/python3.12/dist-packages:/usr/lib/python3/dist-packages:/foss/tools/ngspyce/local/lib/python3.12/dist-packages:/foss/tools/openems/local/lib/python3.12/dist-packages:/foss/tools/yosys/share/yosys/python3:/foss/tools/klayout/pymod
[INFO] Auto-selected local X11.

> [INFO] Waiting until one of the sub-processes stops...

> (xfce4-terminal:13): Gtk-WARNING **: 06:00:04.938: cannot open display: :0
> [INFO] One sub process stopped, exiting... 
```
I tried with and without 'z' option in all '-v' arguments, as well as checking in both Wayland and x11, and ended with the same result. I currently have no idea why, because the script can find the Wayland display/socket, and I'm using XWayland just in case any X11 apps need it

LuighiV, 2025-07-08:
Hi @radityankn 
I faced a similar issue but,  when  I've used an updated image (as suggested in the Glayout docs using the commit stuff) that previously was running in X and I tried to run it using VNC script. I know that this case is different, but I would suggest trying from a fresh pulled image.

Also, when you use the X mode, they should mount some files (sockets) from the host machine. You can check that using:
```
docker inspect iic-osic-tools_chipathon_xserver_uid_1000 
```
and verify two things:
1. The drives are being mounted
```
        "Mounts": [
            {
                "Type": "bind",
                "Source": "/home/luighi/eda/designs",
                "Destination": "/foss/designs",
                "Mode": "rw",
                "RW": true,
                "Propagation": "rprivate"
            },
            {
                "Type": "bind",
                "Source": "/tmp/.iic-osic-tools_chipathon_xserver_uid_1000_xauthority",
                "Destination": "/headless/.xauthority",
                "Mode": "rw",
                "RW": true,
                "Propagation": "rprivate"
            },
            {
                "Type": "bind",
                "Source": "/tmp/.X11-unix",
                "Destination": "/tmp/.X11-unix",
                "Mode": "rw",
                "RW": true,
                "Propagation": "rprivate"
            }
        ],

```

2. The variables for x server are being set:
```
"Env": [
                "DISPLAY=:0",
                "XDG_RUNTIME_DIR=/tmp/runtime-default",
                "XAUTHORITY=/headless/.xauthority",
                "PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin",
                ...
            ],

```

Hope it could help.

amisapta15, 2025-07-08:
Following this issue submission [here](https://github.com/iic-jku/IIC-OSIC-TOOLS/issues/135)

For linux-based system, X forwarding doesn't work very smoothly. I would suggest trying, *either* one of two following approaches, (don't do both)

- Install socat in your host with `apt install socat` or with your favourite package manager 

- instead of Docker Desktop (which runs like s VM) switch to classical Docker CE. 

The first one should be quick to check.

radityankn, 2025-07-13:
@amisapta15 yes, I have installed docker CE. I have verified that helllo world is indeed working. The VNC version of the container is also working

amisapta15, 2025-07-15:
Huh! Thats strange indeed. @MrHighVoltage what do you think?

@radityankn just for a quick check, if you have Docker CE, Can you please try after removing Socat and 'z' modification you mentioned.

Socat was primarily for Docker Desktop. However, automated checks should be there, not to start socat for Docker CE. I am running X script with Docker CE without any modifications or extra libraries.

MrHighVoltage, 2025-07-15:
Hi everyone!

@radityankn Is there anything in the logs for SELinux blocking? Because I never had this issues. The 'z' is probably required anyways, since you are on an SELinux-enforced system.
Also, to test if this issue originates from SELinux, temporarily disable it with running `setenforce 0` as root.

@amisapta15 yes, socat is just used for Docker Desktop on Linux, with Docker CE it will not be used. As you correctly said, there would be an error message anyways it is is required and not found.

Thanks anyways for reporting the issues, this will really help making our container more robust.

MrHighVoltage, 2025-07-19:
This is already very helpful. Thanks! Indeed I guess I need to setup a virtual machine with Fedora to test a similar setup, anyways.

For the logs, there should be `/var/log/audit/audit.log`, which shows SELinux errors, like when it denies the access to something etc... Maybe you can find something specific concerning Docker.

radityankn, 2025-07-19:
Here is the log snippet of what is denied by SELinux, hope it may be helpful to you : 
```
type=AVC msg=audit(1752911122.135:1148): avc:  denied  { connectto } for  pid=231290 comm="xfce4-terminal" path="/run/user/1000/wayland-0" scontext=system_u:system_r:container_t:s0:c254,c564 tcontext=unconfined_u:unconfined_r:unconfined_t:s0-s0:c0.c1023 tclass=unix_stream_socket permissive=1
type=AVC msg=audit(1752911122.216:1149): avc:  denied  { write } for  pid=231290 comm="xfce4-terminal" path=2F6D656D66643A6D75747465722D736861726564202864656C6574656429 dev="tmpfs" ino=77438 scontext=system_u:system_r:container_t:s0:c254,c564 tcontext=unconfined_u:object_r:user_tmp_t:s0 tclass=file permissive=1
type=AVC msg=audit(1752911122.216:1150): avc:  denied  { map } for  pid=231290 comm="xfce4-terminal" path=2F6D656D66643A6D75747465722D736861726564202864656C6574656429 dev="tmpfs" ino=77438 scontext=system_u:system_r:container_t:s0:c254,c564 tcontext=unconfined_u:object_r:user_tmp_t:s0 tclass=file permissive=1
```

closed 2025-07-20 by radityankn

MrHighVoltage, 2025-07-21:
@radityankn How is it fixed now? Can you tell me the solution so we can implement it for everyone?

radityankn, 2025-07-22:
@MrHighVoltage I haven't fixed it yet. I just temporarily disable my SELinux each time I want to use the container, and after that turn it back on after I'm done. The log apparently says that there are some permission issue when connecting to wayland socket and 2 directories (I am not sure what is it). The thing is, I have placed the z option at the wayland socket part of the script (and pretty much everywhere I found -v argument) so it shouldn't have access problem like that