sscs-ose/sscs-chipathon-2025 #44: Chipathon Digital Building Blocks Repo : Team VoidWalkers

radityankn, 2025-07-11:
As mentioned by Mr. Saligane in the zoom meeting room tonight, here is our repo link for progress tracking : 

https://github.com/radityankn/voidwalkers-scandff-gf180mcu

Thank you for your attention, we hope to learn as much as possible from this event

ilmiahmad, 2025-07-13:
[image]

We tried to create a schematic and testbench for a D flip-flop using a Transmission Gate Flip-Flop (TGFF), as recommended by @mithro in an issue on our repository. He referred to a video by @mattvenn as a reference:
https://github.com/mattvenn/flipflop_demo and his video: https://www.youtube.com/watch?v=5PRuPVIjEcs

The current design has been tested with a 100 MHz clock. This design is not final yet. it's just a proof of concept and functional level

RTimothyEdwards, 2025-08-21:
I mistakenly marked this as "approved" on the spreadsheet.  As there is no valid layout for the cell, I have re-marked it as "not approved".  If you have valid layout that has not been pushed to the repo yet, then please do so as soon as possible.

radityankn, 2025-08-22:
Apologies for the late reply, @RTimothyEdwards. Back then, we hadn't done any real layout as we were still having problems with the simulation strategy. Hence, it is the reason why the gds contains only transistor, as its sole purpose is only for area estimation. Since then, we have been doing some layout, but I'm afraid we'll only able to get the first layout pushed at 11 AM ET (10 PM local time, UTC+7). Is it acceptable? Thank you

RTimothyEdwards, 2025-09-01:
Yes, I can incorporate changes into the combined library when they are available.

radityankn, 2025-09-01:
I have checked my power rails and changed them to match the OSU cells (0.7um width). It turned out that my power rail is 0.1um short of the OSU cells

d-m-bailey, 2025-09-04:
@radityankn It is my understanding that @RTimothyEdwards has approved your design for tapeout. I will modify the status on the review sheet. Please prepare a presentation.

d-m-bailey, 2025-09-04:
@radityankn I checked the [design review sheet here](https://docs.google.com/spreadsheets/d/1OnZmy5CrLkoonrmlSXwufpTo5ZXyesfnb08rIsFdK6Q/edit?usp=sharing), and it says you are approved (line 15). Be aware that the sheet is sorted by score and not by team code.

ilmiahmad, 2025-09-05:
# Backup Slides for Layout Review

Here is the link to our layout review slides:  
[Layout Review](https://docs.google.com/presentation/d/138sG9_6nFNde91uJd1VJHh-8p3Jak7Ld7Pb-RK_AmKo/edit?slide=id.g36530e64a14_0_0#slide=id.g36530e64a14_0_0)

d-m-bailey, 2025-11-28:
> Hello @d-m-bailey <https://github.com/d-m-bailey> and @msaligane
> <https://github.com/msaligane> ,
> Please find our slides for Chipathon 2025 Post-Mortem evaluation here
>
> https://docs.google.com/presentation/d/138sG9_6nFNde91uJd1VJHh-8p3Jak7Ld7Pb-RK_AmKo/edit?usp=sharing
>
> Thank you,
> Raditya Eka P

Raditya,

Thanks for preparing the slides - I confirmed the link in the github issue.

We still need:

1. A list of the team members full names and email addresses

Thanks for your understanding
D. Mitch Bailey (not a doctor)
IEEE SSCS TC-OSE