sscs-ose/sscs-chipathon-2025 #50: Team SILICON_BBI1/2 github repo link

silicon-vlsi, 2025-07-12:
https://github.com/silicon-vlsi/chipathon25-SILICON_BBI-sroutk

Please use the above repo as the official repo for both the teams SILICON_BBI1 and SILICON_BBI2 
This repo will consolidate all the cells from the individual contributors from their respective repos into one repo for better tracking by the Chipathon team.

Thank you 
Team SILICON_BBI1/2

silicon-vlsi, 2025-08-06:
adding some relevant fossi chats

silicon-vlsi, 2025-08-11:
@RTimothyEdwards : @d-m-bailey 

Sincere apologies for the delay. Due to school reopenings and member travel, reviews for the past two weeks were missed. Additionally, three members from the SILICON_BBI2 team withdrew. It is proposed to consolidate the remaining seven members into the **SILICON_BBI1** team to deliver **seven standard cells**. All cells have been designed, simulated, and documented in the design review report:

[Google Slides for Design Review](https://docs.google.com/presentation/d/1-PwVBesOc41rMGyMZzqfWGzFfBSO2a17YhIK0xbBM-w/edit?usp=sharing)

A 3-min presentation of the slides can be viewed here:

[3-min Video Presentation](https://zoom.us/rec/share/OGRjbX2RVj9ahWJ8wOJRqyFN8OOf2jaivV1r1lSaEVTwXiixKmu9a4pzv9AdLxja.zR8yUU5UsRYqivPe?startTime=1754899817000)

The project tracker for the project:

https://docs.google.com/spreadsheets/d/1s4SPgW1sIKHpIMEBEdXSYUyiz4REzP_13UfACqfgxFU/edit?usp=sharing

As noted in the presentation, I have the following questions and clarifications:

1. The VDD/VSS were made pins but the current library has globals, is that ok ?
2. As in the current library, we are not creating symbols, hope that’s ok.
3. Is CACE the recommended method for PVT sims ? If yes, please recommend some good tutorials for it.
4. What is the tool used to generate the LEF files ?
5. What is the expected date for submitting the final library package ie. lib/lef/sch/etc
6. And what are the files (lib/lef/sch/etc) expected to be submitted for the library.
7. Currently we are not able to install CharLib on the docker and having issues with ngspice on a separate installation. Any help on that will be appreciated.

Apologies again for the delay, hope we can continue.

RTimothyEdwards, 2025-08-11:
(1) I suppose I didn't give that enough thought.  Schematics for use in LVS and simulation _should_ have power pins, and the corresponding symbol used for testbench simulations _should_ have power pins, but symbols used for placing standard cells in schematics traditionally _don't_ have power pins.  I think two symbols are needed, one with power pins and one without.  The one without would be a "primitive" and would not correspond to the schematic;  it would be used in a schematic as a black-box entry, with the assumption that the netlists for the entire standard cell library are included in a top-level simulation.

(2) Symbols will be needed but can just be pulled from the xschem set for sky130, which is pretty comprehensive, with pin names changed as appropriate.

(3) CharLib is recommended, although we have not discussed how to set it up for PVT corners, and the existing liberty files for the OSU library only represent the typical corner.  There really needs to be at least "min" (fast/high/cold) and "max" (slow/low/hot) liberty files in addition to "nom" (typical/typical/typical).  Since "min" and "max" libraries do not currently exist for the OSU library, this is not a requirement for Chipathon.

(4) I generate LEF files from magic using "lef write".

(5) Next week would be good.  It will be hard to prepare anything for a September tapeout unless standard cells are done and available for use early.

(6) Schematic (.sch), symbol (.sym), layout (.gds), LEF (.lef), SPICE (.spice), liberty (.lib).  A .mag file if you did the layout in magic.

(7) You may need to get help from the OSU team for CharLib support.  I have installed the PDK natively and was able to run CharLib according to instructions (pip install into a python virtual environment).

silicon-vlsi, 2025-08-21:
@d-m-bailey @RTimothyEdwards : 
Now all the cells in the repo are DRC/LVS clean for the team D15_SILICON_BBI+2 repo :

`https://github.com/silicon-vlsi/chipathon25-SILICON_BBI-sroutk/`

- The `.lib` file is created from the layout extracted file with parasitics except for `12T NOR3_1X` for which charlib failed for the extreme corner of `10ns` slew rate and `1.5pF` load. Right now the `.lib` for that cell is from the schematic netlist. I will manually do a spice sim for that corner an update that cell soon.
- `GDS/LEF` files are also created in put in the repo.
- We have adhered to the same directory structure as the OSU library.
- Testbenches, charlib files, extraction scripts, etc can be found in the `tb_digital` folder

Please take a look and let us know any changes or info needed for those six stdcells