sscs-ose/sscs-chipathon-2025 #51: Time Transcenders proposal link

Landflier, 2025-07-12:
https://github.com/Landflier/Chipathon_2025_gLayout/blob/main/doc/TimeTranscenders_Proposal.pdf

amisapta15, 2025-07-20:
A resource shared by @peterkinget , might be helpful. A detailed response to the full proposal will follow shortly. 

https://www.ee.columbia.edu/~kinget/EE6350_S24/AMRADIO/ic_design.html

msaligane, 2025-08-08:
@Landflier I am not sure. Maybe you can share more of your progress and how you are structuring your design.

shoebNTU, 2025-08-11:
thanks for the reply @msaligane . We have reduced the frequency to 100 MHz and are following the schematic proposed by Prof. Kinget. Here is our tracking sheet - (WIP) https://docs.google.com/spreadsheets/d/12n58GIiJvpNae0BYuF45ltOiLeXCqi4JbQlBEFmj9gw/edit?usp=sharing We are working on our final report in the format request by Mitch. 

My team-mate is working on the schematic finalization and had the following questions - 

- When they said that we should include an on-chip bias, would that mean including a bandgap referenece? What would be the    problem of just having a dc+ac bias from the chip pads? (Can we just reuse D15 XDAC team- xjlee, they already have the block, need to change for 50uA)
- Can our impedance matching network be off-chip?
- With on-chip, off-chip components, is there any guidance to go about our test set-up?
- For 3-6K Ohm resistors, would the ppolyf_u be good? What is the difference between the resistor devices?

On the glayout front, I am exploring to build an LLM powered MCP server for glayout. Thanks a lot for your guidance.

peterkinget, 2025-08-12:
- On-chip bias is voltage bias for the various devices. That bias is very often dependent on the VT of the transistors so one external bias for all chips does not work. But if you run out of time, you can do an external bias with a Bias-Tee. 
- I would not put a bandgap reference on your chip -- too much added risk for very little added benefit; just bring in one reference current from outside and mirror that to the various blocks.
- How are you going to test this mixer? Draw a full application schematic that includes the external generators (and bias) with the correct impedances and the measurement equipment (likely a spectrum analyzer which is 50 Ohm, or a high speed scope, which is likely also 50 Ohm). Now you need to worry how to drive the 50 Ohm load from your chip, or add buffers on your PCB; both are not trivial and come with their own challenges. 
- I am not familiar with the resistor options (yet). The choice of resistor type will depend on its function in the circuit.

LuighiV, 2025-08-12:
Hi @shoebNTU, @peterkinget 

Hope you don't mind, but I'd like to comment a little about this point:

> * For 3-6K Ohm resistors, would the ppolyf_u be good? What is the difference between the resistor devices?

In our repository we have done a characterization for the resistor including temperature and process variation https://github.com/AutoMOS-project/AutoMOS-chipathon2025/tree/main/designs/libs/tb_analog/tb_resistor. According to our results, essentially the difference comes with the TCs, sheet resistance value and temp-dependence linearity:

[image]

Some insights from it:
- **Metal**, (rm1,rm2,rm3) in general has the lowest sheet resistance value (~0.1 Ohm), and positive linear TC.
- **Salicide Polysilicon**, (ppolyf_s, npolyf_s) the next in sheet resistance value (~20 Ohm) and positive quasi-linear TC.
- **Unsalicide Polysilicon,** (ppolyf_u, npolyf_u) with a mid-resistance value (~400, 500 Ohm) and negative quasi-linear TC.
- **Unsalicide Polysilicon with high resistance layer** (ppolyf_u_1k) with higher resistance (~1.2k Ohm) and negative non-linear TC.
- **Wide N+/P+ difussion** (nplus_u, pplus_u) with a mid-resistance value (~90 Ohm for N+, and 300 Ohm for P+) and quasi-linear positive TC.
-  **N Well** (nwell) with a high resistance value (~1.4k Ohm) and positive non-linear TC.

Then, I think one criterion to choose your resistance is which is the target value. If it is high like hundred of kOhms to MOhms, probably a ppoly_u_1k might be better, but if it is just a few kOhms the ppoly_u might work as well, as you are planning to use. Just be aware of the sign of the TC and how it might impact inside your circuit.

Hope this could help.

Regards,
Luighi

d-m-bailey, 2025-08-13:
@Landflier 
Please update the first comment in this issue according to the instructions here #101.

Be sure to include a progress tracker so that we can judge the likelihood of completion.

d-m-bailey, 2025-08-19:
@Landflier All links should be in the first comment of this github issue. I will remove the links from the review sheet.

JuanMoya, 2025-08-28:
Dear Team leader, I hope you are well.

We are starting the final phase of the 2025 Chipathon, which is the Integration phase.

Follow the  link to a document with all the different tasks that need to be completed in the next days and weeks.

https://docs.google.com/spreadsheets/d/1cUdG603xd2MkME8X6REkCbpfBFh-e17oDE7eaJLUmfE/edit?usp=sharing

I will create a folder for milestone 2 and let you know where to place all the items requested.

Best regards

Landflier, 2025-08-29:
Our team will need to implement the secondary ESD protection cell mentioned in the presentation this Friday (29.08.2025), since we have 6 FET's with their gates directly tied to a PAD.

JuanMoya, 2025-08-29:
Hi @Landflier , will you be interested in designing this secondary ESD structure?