sscs-ose/sscs-chipathon-2025 #55: Digital Building Blocks - Team Setup & Hold

diadatp, 2025-07-13:
Our proposal addresses a gap in the OSU 12-track standard cell library by designing and characterizing a 4-to-1 multiplexer at single and double drive strength.

https://github.com/diadatp/sscs-chipathon-2025-team-setup-and-hold/blob/main/proposal.md

Team Members:
@diadatp
@kavsdev
@gitshriram
@recursivetree

recursivetree, 2025-08-11:
Here is the requested link to proposal slides: https://docs.google.com/presentation/d/e/2PACX-1vQ9X2YuV-HSiZHVjvDTaz7JzXOTwWBogRJj-HR7-0yTNbbXNtv4UYqzkiuTKmifFw/pub?slide=id.g36e4c9c564b_113_0
However, we changed form a MUX4 to a AOI222 cell because multiple teams are doing a MUX4, so it is not entirely relevant. Please check our updated proposal in the github repo.

d-m-bailey, 2025-08-13:
@diadatp @recursivetree 
Is it possible to update the first issue to include links all the requirements listed here #101.