sscs-ose/sscs-chipathon-2025 #56: CP25:Team-Mahowalders-Design-Regulated-Cascoded-Current-Mirror

Sud-ana, 2025-07-13:
Team Mahowald-ers: PR for adding Regulated Cascoded Current Mirror block. 
This block will add a Cascode regulator block from this team's other PR for enhanced voltage swing.
Ref: DOI: 10.1109/AE54730.2022.9920096
This is an initial Commit, and a proposal (with description) of this PR can be found in this link or in the README of our folder in the composite block.

Docs Proposal link (Viewing option open only for Comments):
https://docs.google.com/document/d/1FZdyJ7MXu66QQ6_0kkzPwPryskKX2VRuOPmt5tDAb5Y/edit?usp=sharing

Repo Proposal:
https://github.com/Sud-ana/gLayout-mahowalders/blob/main/blocks/composite/regulated_cascoded_cmirror/design_proposal/README.md

Pull Request:
https://github.com/ReaLLMASIC/gLayout/pull/32

alfadi01, 2025-08-01:
Please find our progress tracker sheet here : https://docs.google.com/spreadsheets/d/1k9veHdDh-e81fpdIpsUblsUgEMs8aU_Jj_eBLXmVOYY/edit?usp=sharing

d-m-bailey, 2025-08-13:
@Sud-ana @alfadi01 
Is it possible to update the first issue to include links to all the documents listed here #101

alfadi01, 2025-08-15:
[TOP.md](https://github.com/user-attachments/files/21798178/TOP.md)

d-m-bailey, 2025-11-24:
Just wanted to let you know that your block passed LVS with only property errors in the `io_secondary_5p0` cell.