sscs-ose/sscs-chipathon-2025 #58: MOSbius Track: Low Noise Anomaly Team

mthudaa, 2025-07-13:
Here, our repository: https://github.com/mthudaa/sscs-chipathon2025-LNA
And here our proposal: https://github.com/mthudaa/sscs-chipathon2025-LNA/blob/main/docs/our_proposal.md

amisapta15, 2025-07-20:
This is a copy/pasted Feedback from @peterkinget (that is originally shared in Element). The aim is to preserve and track the feedback and subsequent changes integrated into the design. **Please note: this feedback is shared just after the 4th Weekly Meeting. Further Feedback for the full proposal will be shared in due course.** 

# [MOSbius Track] Feedback for Low Noise Anomaly

**M. Taufiqul Huda**

- I looked at your slides and full proposal — nice write-up.  
- This is a very ambitious proposal for the time available.  
- What is the architecture of the switch matrix? 16 buses × ??? device terminals?  
- It's interesting you are adding LE. What advantage does that offer over an FPGA on the PCB?  
  - How will you map a digital circuit on your mini-FPGA? By hand or do you have a tool in mind?  
  - What kind of digital functions do you plan to realize with your LEs?  
  - What is the schematic of the LE?  
- How will you realize the RAM?  
  - Do you have experience with building mixed analog/digital chips? The design and verification flow can be daunting for first-time users.  
  - How about using a scan chain — would that work?  
- I would advise connecting with the other teams to share blocks and collaborate.  
- For the analog sections, take an approach similar to what I just suggested for the FPTA team above.

peterkinget, 2025-08-09:
@mthudaa 

some feedback on your slides:
- please make sure to make a progress tracker
- please estimate the number of pins you need and the area

We will share how to share that information efficiently for our go/no-go review in the coming week.

d-m-bailey, 2025-08-13:
@mthudaa 
Please update the first comment in this issue according to the instructions here #101.

Be sure to include a progress tracker so that we can judge the likelihood of completion.

d-m-bailey, 2025-08-14:
@mthudaa Thanks for preparing the gds data. You will need to create a pull request to add the data according to the instructions [here](https://github.com/sscs-ose/sscs-chipathon-2025/tree/main/resources/Integration/Milestone_8_15_2025/MOSbius/M6_LowNoiseAnomaly).

d-m-bailey, 2025-08-14:
@mthudaa There are more detailed instructions [here](https://github.com/d-m-bailey/sscs-chipathon-2025/blob/dry-run-instructions/resources/Integration/Milestone_8_15_2025/README.md).

JuanMoya, 2025-08-28:
Dear Team leader, I hope you are well.

We are starting the final phase of the 2025 Chipathon, which is the Integration phase.

Follow the  link to a document with all the different tasks that need to be completed in the next days and weeks.

https://docs.google.com/spreadsheets/d/1cUdG603xd2MkME8X6REkCbpfBFh-e17oDE7eaJLUmfE/edit?usp=sharing

I will create a folder for milestone 2 and let you know where to place all the items requested.

Best regards

mthudaa, 2025-12-04:
Dear @d-m-bailey 

I attach the postmorthem review

[Chipathon 2025 Postmortem - M6.pptx.pdf](https://github.com/user-attachments/files/23945346/Chipathon.2025.Postmortem.-.M6.pptx.pdf)

Thank you,

Huda