sscs-ose/sscs-chipathon-2025 #60: RF MOSbius - PLL MOSbius Full Proposal

AndrewwChon, 2025-07-13:
https://github.com/AndrewwChon/PLL_MOSbius/tree/main/proposal

amisapta15, 2025-07-20:
This is a copy/pasted Feedback from @peterkinget (that is originally shared in Element). The aim is to preserve and track the feedback and subsequent changes integrated into the design. **Please note: this feedback is shared just after the 4th Weekly Meeting. Further Feedback for the full proposal will be shared in due course.** 

# [MOSbius Track] Feedback for RF_MOSbius

**andrewchon**

- Please submit a link to your repo and your full proposal on GitHub and include a work distribution and schedule.

xavierjlee0, 2025-08-08:
Hey! I saw your question on how to speed up simulations for your block. IIRC, specifying only a handful of nodes in your .save command (as opposed to .save all) has helped dramatically in other spice simulation software I have used. Maybe you'd see some benefit? Let me know!

AndrewwChon, 2025-11-28:
Link to Post-Mortem Slides below.

[[Post-Mortem Slides.pdf](https://github.com/user-attachments/files/23828252/Post-Mortem.Slides.pdf)
](https://github.com/AndrewwChon/PLL_MOSbius/blob/main/Post-Mortem%20Slides.pdf)