sscs-ose/sscs-chipathon-2025 #65: Mosbius Track - Team Nirvana

chennakeshavadasa, 2025-07-14:
https://github.com/chennakeshavadasa/SSCS-Chipathon-2025-Team-Nirvana

peterkinget, 2025-08-09:
@chennakeshavadasa  

some feedback on your slides:
- please make sure to make a progress tracker
- please estimate the number of pins you need and the area

We will share how to share that information efficiently for our go/no-go review in the coming week.

d-m-bailey, 2025-08-13:
@chennakeshavadasa 
Please update the first comment in this issue according to the instructions here #101.

Be sure to include a progress tracker so that we can judge the likelihood of completion.

d-m-bailey, 2025-08-15:
@snighdhasarali thanks for adding the progress tracker link. Please review the instructions here #101  for other necessary data.