sscs-ose/sscs-chipathon-2025 #66: Chipathon Analog/AI Track project: AutoMOS team

LuighiV, 2025-07-14:
Hi all,

This issue is to post the information about our proposal for the Chipathon 2025. 
We are in the Analog/AI track.

 - **Repository**: https://github.com/AutoMOS-project/AutoMOS-chipathon2025
 - **Proposal**: https://github.com/AutoMOS-project/AutoMOS-chipathon2025/blob/main/docs/AutoMOS-project_Proposal.pdf

Thanks,
AutoMOS team

LuighiV, 2025-08-01:
## Meeting Feedback (2025/08/01)

Hi @msaligane  and @peterkinget  thanks for your comments during today's meeting. These are the questions we have captured:

- PK: VCO requires a buffer for the output. 
   Ans: We are going to add it using the pcell suggested by M. Saligane

- MS: are you using devices of 3.3V/5V? 
   Ans: Yes, our main voltage is 5V it powers the bangap, the biasgen and LDO, for all of them we are using 5v0 devices. It generates the 1.8V subdomain that powers the VCO (this hierarchy uses the 3v3 devices).

- MS: idenfify which subcircuits can be used from Glayout. 
  Ans: yes, we plan to also add as a annotations to the schematic itself and do a review on the galyout repo if they exists or not.

Thanks, please, let us know if there is somethig missing.

AutoMOS team

msaligane, 2025-08-08:
@LuighiV I would encourage you to open a PR to the glayout repo with your new pcells & blocks (and follow the structure of folders there).

Regarding your questions:
 - 5v devices are available. It should be in the doc. @d-m-bailey can we share the list of devices we sent to GF?
 - I am not sure how to answer this question. How many IO pads you need
 - I need more context for what gpios you mean
 - You ll need to create a pcell for it. 
 - Design parameters to care about:
Input voltage range (Vin,min → Vin,max)
Output voltage (Vout)
Max load current (Iout,max)
Dropout voltage (VDO @ Iout,max)
Output accuracy (line + load + temp)
Line regulation
Load regulation
Noise (PSSR)
There are many other parameters to check. Have you designed an LDO before?

d-m-bailey, 2025-08-09:
@peterkinget It was in the 180MCU_DeviceSelector chipathon.xlsx that was submitted for the September tapeout. I don't think it has been made public.

RTimothyEdwards, 2025-08-09:
The magic tech file also has a list of supported device, but with the correct device names for the open PDK:
```
# Devices:
# All devices except diodes are modeled as subcircuits

# device list:
#  nfet_03v3            mosfet
#  pfet_03v3            mosfet
#  nfet_06v0            mosfet (thick oxide)
#  pfet_06v0            mosfet (thick oxide)
#  nfet_05v0            mosfet (thick oxide, short gate)*
#  pfet_05v0            mosfet (thick oxide, short gate)*
#  nfet_06v0_nvt        mosfet (native Vt)
#  nfet_03v3_dss        mosfet (unsalicided drain)
#  pfet_03v3_dss        mosfet (unsalicided drain)
#  nfet_06v0_dss        mosfet (unsalicided drain, thick oxide)
#  pfet_06v0_dss        mosfet (unsalicided drain, thick oxide)
#  nfet_10v0_asym       LDNMOS (extended drain)
#  pfet_10v0_asym       LDPMOS (extended drain)
#
#  diode_nd2ps_03v3     diode  (N+/pwell)
#  diode_pd2nw_03v3     diode  (P+/nwell)
#  diode_nd2ps_06v0     diode  (N+/pwell, high voltage)
#  diode_pd2nw_06v0     diode  (P+/nwell, high voltage)
#  diode_nw2pw_03v3     diode  (nwell/pwell)
#  diode_nw2pw_06v0     diode  (nwell/pwell, high voltage)
#  diode_dnw2pw         diode  (pwell/dnwell)
#  diode_dnw2ps         diode  (dnwll/substrate)
#  sc_diode             diode  (Schottky)
#
#  pnp_WxL              BJT    (10x10, 5x5, 0.42x10, 0.42x5 emitter sizes)
#  npn_WxL              BJT    (10x10, 5x5, 0.54x16, 0.54x8, 0.54x4, 0.54x2)
#
#  npolyf_u             resistor (N+ poly, unsalicided)
#  ppolyf_u             resistor (P+ poly, unsalicided)
#  ppolyf_u_1k          resistor (high res resistor)*
#  ppolyf_u_1k_6p0      resistor (high res resistor, high voltage)**
#  nplus_u              resistor (N+ diffusion, unsalicided)
#  pplus_u              resistor (P+ diffusion, unsalicided)
#  npolyf_s             resistor (N+ poly, salicided)
#  ppolyf_s             resistor (P+ poly, salicided)
#  nplus_s              resistor (N+ diffusion, salicided)
#  pplus_s              resistor (N+ diffusion, salicided)
#  nwell                resistor (N-well resistor)
#  rm1                  resistor (metal1)
#  rm2                  resistor (metal2)
#  rm3                  resistor (metal3)
#  rm4                  resistor (metal4)
#  rm5                  resistor (metal5)
#  tm6k                 resistor (top metal, 0.6um thick (standard))***
#  tm9k                 resistor (top metal, 0.9um thick)
#  tm11k                resistor (top metal, 1.1um thick)
#  tm30k                resistor (top metal, 3.0um thick)
#
#  cap_nmos_03v3        mosfet   (source-drain tied)
#  cap_pmos_03v3        mosfet   (source-drain tied)
#  cap_nmos_06v0        mosfet   (source-drain tied, high voltage)
#  cap_pmos_06v0        mosfet   (source-drain tied, high voltage)
#  cap_nmos_03v3_b      mosfet   (n-varactor)
#  cap_pmos_03v3_b      mosfet   (p-varactor)
#  cap_nmos_06v0_b      mosfet   (n-varactor, high voltage)
#  cap_pmos_06v0_b      mosfet   (p-varactor, high voltage)
#
#  cap_mim_2f0fF        capacitor (MiM)**,***
#
#  efuse                fuse

# *Note that the "5V" and "6V" devices are the same device and have the same
#  device model.  However, a slightly shorter gate length is allowed when
#  operating at 5V.  The GF standard cell sets use the 5V length.  Magic
#  will extract the shortest gate length as a "5V" type and everything else
#  as "6V", which should allow for clean LVS when using the standard cells.
#
# **Note that there are multiple mutually exclusive process options for the
#  high sheet rho resistor and MiM capacitor.  This tech file assumes the
#  options 1kOhm/sq for the resistor and 2fF/um for the MiM capacitor.
#
# ***The top metal may be any of metal3 to metaltp, depending on the stackup
```

msaligane, 2025-08-11:
@LuighiV Looking forward to your update soon. Thanks!

LuighiV, 2025-08-11:
Hi @msaligane 

Thanks for your feedback. Please, find below some comments/updates regarding that:

> [@LuighiV](https://github.com/LuighiV) I would encourage you to open a PR to the glayout repo with your new pcells & blocks (and follow the structure of folders there).

Indeed, we already created a fork for the glayout repository, which is a submodule of the repository pointed above:
https://github.com/AutoMOS-project/gLayout-AutoMOS-cp25/tree/2eb1cc04a6f6a15bad73e701f565984372006175 inside the `designs` folder: https://github.com/AutoMOS-project/AutoMOS-chipathon2025/tree/main/designs

We cannot create the PR because don't have a commit yet in the repository. Actually, we are not sure how to structure the blocks, composites inside the glayout repository.

> 
> Regarding your questions:
> 
> * 5v devices are available. It should be in the doc. [@d-m-bailey](https://github.com/d-m-bailey) can we share the list of devices we sent to GF?

Thank you for confirming, and thanks @d-m-bailey for sharing that table. It is really helpful.

> * I am not sure how to answer this question. How many IO pads you need

This is related to the information posted in the https://github.com/sscs-ose/sscs-chipathon-2025/tree/main/resources/Integration page. I was wondering if the chip size available for each team is the one posted there (2040um x 2040um).

In our case, originally, we were planning to use 11 IO pads, however we increased following last meeting @peterkinget suggestion to include some overrides for the internal references. Now, we are considering 21.

> * I need more context for what gpios you mean

It is linked to the previous question. I mean, is it supposed that we are going to use the guard ring posted there? If that is the case, is it expected that we are going to place the required gpios (or IO cells) for the inputs/outputs there? 

> * You ll need to create a pcell for it.

Ok, we are looking at the repository and checking the primitives for fets, is it expected for us to create something similar?

> * Design parameters to care about:
>   Input voltage range (Vin,min → Vin,max)
>   Output voltage (Vout)
>   Max load current (Iout,max)
>   Dropout voltage (VDO @ Iout,max)
>   Output accuracy (line + load + [...]

Thanks for the information. In fact, my teammate @jvilcap, who is designing it, asked the question. He hasn't designed it before, but has experience by his thesis on a PM unit. I also have experience in PM modules because of my job and looked at LDOs previously. Our question was more related to numerical values, that might be considered acceptable for the LDO.
We are also working in an SCM for this block. I think it will be more helpful to explain this point.

LuighiV, 2025-08-13:
Hello Prof. @msaligane,
Thanks for the feedback during yesterday's meeting. I've added the link to the GitHub project we are using to track the activities and other information to the first message, as indicated by @d-m-bailey. There are still some items that are WIP and we hope can finish before the deadline.

Thanks,
Luighi

JuanMoya, 2025-08-28:
Dear Team leader, I hope you are well.

We are starting the final phase of the 2025 Chipathon, which is the Integration phase.

Follow the  link to a document with all the different tasks that need to be completed in the next days and weeks.

https://docs.google.com/spreadsheets/d/1cUdG603xd2MkME8X6REkCbpfBFh-e17oDE7eaJLUmfE/edit?usp=sharing

I will create a folder for milestone 2 and let you know where to place all the items requested.

Best regards

d-m-bailey, 2025-11-22:
@LuighiV probing the power nets as suggested above, shows some taps that might not be connected. The ntaps are more important than the ptaps, I think.

[image]
[image]

LuighiV, 2025-11-22:
Hi @d-m-bailey 

Thank you so much for finding those errors, I think the substrate taps not connected were because we added some in the middle to pass DRC but I forgot to connect. I'm fixing them right now. 

Regarding the issue with the connection to tieh, I'm sorry, that was my mistake, I've totally missed. When I checked the trace net, I didn't notice it didn't complete the connection till the internal VDD net. I'm going to fix that as well. Unfortunately, that might mean that the circuit may not work  😢

RTimothyEdwards, 2025-11-22:
@d-m-bailey :  Don't overstate the issue with magic dropping vias.  It requires a very specific and pathological geometry to achieve that.

d-m-bailey, 2025-11-23:
@LuighiV It is fine to have ntaps and ptaps connected to nets other than `vdd` and `vss` as long as they are separated.

I haven't received an answer about whether or not deep nwell needs a surrounding nwell/ntap guard ring.

Other issues 
1. `test_tg` layout and schematic are entirely different circuits. For LVS, I recommend making sure that layout and schematic blocks with the same name refer to the same circuit. netgen will automatically flatten mismatched circuits, but I don't think klayout will.
2. In `bandgap` all the poly resistors have a `VDD` connection in the schematic, but in the layout they are over psubstrate which is `VSS`.

There have been some recent updates to the magic tech files for gf180mcu. Checking the differences now.

LuighiV, 2025-11-23:
Hi @d-m-bailey 

Thank you for your feedback. Regarding the issues you commented:
- For the test_tg my teammate opted to join all the individual tg_cells and related connections in a macro block, however he used the same name. I'm going to update the schematic to reflect what we have there.
- Regarding the band gap, yes, I have noticed that yesterday and also updated the schematic when I did the updates for the layout.

Thanks,
Luighi

LuighiV, 2025-11-24:
Hi @d-m-bailey 

I was trying to check the blocks after I updated the PDK, but found the following error, have you faced that previously?

[image]

d-m-bailey, 2025-11-24:
@LuighiV I think I'm seeing a connection difference between the schematic and the layout in the `bandgap` circuit.

In the schematic the output of 16 parallel bipolar transistors `vbjt_ptat[0]` is connected to 6 resistors in series whose output is `vp`, while in the layout it is connected to 54 resistors in series whose output is `VBG`. Is this swapped with `vbjt_bg[0]`?