sscs-ose/sscs-chipathon-2025 #68: XDAC Team: MOSbius Style Class-D Amplifier

xavierjlee0, 2025-07-14:
Our proposal can be found in our group repository here:
https://github.com/xavierjlee0/xdac_chipathon2025/blob/main/docs/XDAC%20Official%20Proposal.pdf 

Alternatively, a PDF copy has been uploaded here:
[XDAC Official Proposal.pdf](https://github.com/user-attachments/files/21211480/XDAC.Official.Proposal.pdf)

amisapta15, 2025-07-20:
This is a copy/pasted Feedback from @peterkinget (that is originally shared in Element). The aim is to preserve and track the feedback and subsequent changes integrated into the design. **Please note: this feedback is shared just after the 4th Weekly Meeting. Further Feedback for the full proposal will be shared in due course.** 

# [MOSbius Track] Feedback for XDAC

**xjlee**

- Class D amplifiers are great circuits to learn mixed signal design; they have analog & feedback, digital, and power all in one circuit...  
- And they can be demonstrated in audio applications.  
- What output power are you targeting?  
- Start early on the layout of the output devices, those are challenging to design and lay out.  
- Please submit your full proposal and include a timeline, an application diagram, and specification targets.  
- https://vlsidesignlab.org has several class-D amplifier design examples; here are two:  
  - https://www.ee.columbia.edu/~kinget/EE6350_S23/Group03Website/system_overview.html  
  - https://www.ee.columbia.edu/~kinget/EE6350_S14/ClassD_SKSS/systemOverview.html  
- It would be best to include the sawtooth generator on the chip with an option to bring an external sawtooth in for testing.  
- There are very few cells you can synthesize with Verilog in the Class D core; they are designed full custom typically.

peterkinget, 2025-08-09:
@xavierjlee0 

some feedback on your slides:
- please make sure to make a progress tracker
- please estimate the number of pins you need and the area

We will share how to share that information efficiently for our go/no-go review in the coming week.

peterkinget, 2025-08-15:
@xavierjlee0 

I included your layout into the top-level padframe but it looks strange ... Compare it to the other ones ... It seems your database units could be off. Not sure actually. 

[image]

xavierjlee0, 2025-08-15:
Thank you for the update! I agree, something looks off. I'll look into it.

RoyceRichmond, 2025-08-16:
One way to avoid errors in the database units could be to export the cells as a library. The GDS files for each cell can be exported as a library in KLayout and then instantiated in a new layout. This ensures that everything remains at the correct size and prevents the cells from being separated. (When I previously imported the GDS, each NMOS was imported as individual layers.)

Christopher-Amankwaa, 2025-08-16:
@peterkinget, @d-m-bailey  & @RoyceRichmond I believe the layering issue came from a layering offset that klayout places when a gds file is instantiated. We are working to fix this along with the the dbu.

Christopher-Amankwaa, 2025-08-22:
> > [@peterkinget](https://github.com/peterkinget) There was a comment on the Mosbius no/no-go spreadsheet that mentioned the importance of pvt in the non-overlap generator. I already tested high and low temperatures' effects on the delay time, so my question is, what additional analysis would you like to see?
> 
> Also study process variations and voltage variations; e.g. SS + hot + low supply and FF + cold + high supply and everything typical is a min. set.

@peterkinget Hello, Peter. I am working on finalizing and combining all of XDAC's simulations, and to that point, I am having trouble finishing the PVT analysis for the deadtime generator. I have been able to simulate high and low temperatures, as well as high and low voltages, but I am having trouble accessing the electron/hole mobility parameter to run fast-fast corner and slow-slow corner calculations in Xschem. Do you know how I could go about this, or have any resources I could use to help figure out the best method?

Christopher-Amankwaa, 2025-08-22:
Changing typical to ff and ss in the model callout seemed to work. Thanks, Peter.

JuanMoya, 2025-08-28:
Dear Team leader, I hope you are well.

We are starting the final phase of the 2025 Chipathon, which is the Integration phase.

Follow the  link to a document with all the different tasks that need to be completed in the next days and weeks.

https://docs.google.com/spreadsheets/d/1cUdG603xd2MkME8X6REkCbpfBFh-e17oDE7eaJLUmfE/edit?usp=sharing

I will create a folder for milestone 2 and let you know where to place all the items requested.

Best regards