sscs-ose/sscs-chipathon-2025 #75: MOSbius Track - Spikcore

RoyceRichmond, 2025-07-15:
# Team Spikcore # 
We are the team spikcore, we consist of 3 members

Royce Richmond [team leader]
Rosendo Valdés
Oscar Islas

The repository ca be found here [Repo](https://github.com/RoyceRichmond/Mosbious_2025_spiking4all)

The full proposal can be found in here [Presentation SSCS 2025](https://docs.google.com/presentation/d/1-4U3aoSUD9l36EbWa95a9lBhWrAFakjFk_HbliW-nGI/edit?usp=sharing)

amisapta15, 2025-07-20:
This is a copy/pasted Feedback from @peterkinget (that is originally shared in Element). The aim is to preserve and track the feedback and subsequent changes integrated into the design. **Please note: this feedback is shared just after the 4th Weekly Meeting. Further Feedback for the full proposal will be shared in due course.** 

# [MOSbius Track] Feedback for SpikCore

**Royce Richmond**

- Please submit a link to your repo and full proposal on GitHub.  
- Interesting chip block diagram and application area; nice to see some circuit simulations also.  
- Can you provide a diagram mapping one application (neural network) to your chip? What kind of NN can you support with one chip? Or, will you need multiple?  
- You propose to extend the shift register with digital control bits, what will they control specifically?  
- For the switch matrix design, reach out to other teams to collaborate to save design time.

amisapta15, 2025-07-20:
I would recommend drawing a diagram or a schematic that takes the application diagram and shows how you wire it up with the components on your chip. That is a good way to make sure you are not missing anything and also will start giving you a sense of which parts of the chip are most critical. 

You should start with those to build a "minimum viable product" and then expand from there given there is not much design time in this project.

RoyceRichmond, 2025-07-21:
the proposal was updated with the building blocks annd how it would be connected

RoyceRichmond, 2025-08-01:
Updated the testbench simulation to include power consumption for the zigzag power scheme, under designs/libs/tb_synapse_pc

added gds library for the abstracts form of the standard cells under /designs/klayout/gf180mcu_fd_sc_mcu9t5v0_abstract.gds

RoyceRichmond, 2025-08-04:
Good night. @peterkinget 
I have been working on the dff layout. While using the standard cell library the lvs fails (because it imports the complete latch layout which is about 16 mos and more netlists ) and it is a mismatch between that and the schematic for the dff which only has about 7 netlist.

There's another library on the lef folder that has an "abstract" representation of the standard cells which only includes the p/n well, metal 1 and boundaries I ported part of that to a compatible gds for klayout (although it has some caveats ) https://github.com/RoyceRichmond/Mosbious_2025_Spikcore/blob/main/designs/klayout/gf180mcu_fd_sc_mcu9t5v0_abstract.gds

d-m-bailey, 2025-08-05:
Questions from 8/5 office hours.

Problems with klayout LVS.

@RoyceRichmond Maybe we can continue the LVS conversation here.

RoyceRichmond, 2025-08-19:
We'll probably need more than 22 pins but we'll be working on that issue.
Also we have been talking with OnChip team to see whether we can have a unified project, my question would be about the available pinout (would we have the same 22 pins for both projects or could we have more pins ?)

JuanMoya, 2025-08-28:
Dear Team leader, I hope you are well.

We are starting the final phase of the 2025 Chipathon, which is the Integration phase.

Follow the  link to a document with all the different tasks that need to be completed in the next days and weeks.

https://docs.google.com/spreadsheets/d/1cUdG603xd2MkME8X6REkCbpfBFh-e17oDE7eaJLUmfE/edit?usp=sharing

I will create a folder for milestone 2 and let you know where to place all the items requested.

Best regards