sscs-ose/sscs-chipathon-2025 #76: MOSBius Track - Lumut Bius

candrasengkala, 2025-07-16:
Attached on this issue is our group's proposal. Thank you!

[Proposal](https://github.com/candrasengkala/MOSBius-style-Modular-Data-Conversion-Chip/blob/main/proposal/proposal.pdf)
[GitHub Repository](https://github.com/candrasengkala/MOSBius-style-Modular-Data-Conversion-Chip/)

peterkinget, 2025-07-26:
Thanks for your proposal. 
- generating clocks of chip can sometimes be more challenging than on chip; how will you generate them on the PCB?
- add specs for your ADC
- work out a more detailed proposal!
- add a work distribution
- add a timeline
- start designing .... you need to get into the nitty gritty to understand the challenges
- I would just focus on an ADC, that by itself will be challenging given your preparation.

d-m-bailey, 2025-08-13:
@candrasengkala 
Please update the first comment in this issue according to the instructions here #101.

Be sure to include a progress tracker so that we can judge the likelihood of completion.