sscs-ose/sscs-chipathon-2025 #77: Digital Building Blocks - BitSamaMirai

kana820, 2025-07-16:
Hi, we are BitSamaMirai on the Digital Building Blocks track. We are going to work on the XOR3 cell for GF180MCU OSU (9 tracks).
Team members:
- Keertha N <@keerthanamatrix.org:matrix.org>
- Roy Victor Roberto <@roirobertooo:matrix.org>
- Reinhard Iven <@ri.winata:matrix.org>
- Kana Takizawa <@kana38:matrix.org>

Please find our proposal in our github repository [here](https://github.com/reinvalind/SSCS-Chipathon-2025---bitsamaMirai). We'd really appreciate any feedback.

RTimothyEdwards, 2025-08-07:
Please organize your repository, putting schematics, symbols, and testbenches into an `xschem/` folder.  Please separate the cell schematic from the testbench schematics.  Please honor the naming convention for cells, so a low-drive output cell would be `xor3_1`.  Do not used mixed case in cellnames.

d-m-bailey, 2025-08-13:
@kana820 
Please update the first comment in this issue according to the instructions here #101.

Be sure to include a progress tracker so that we can judge the likelihood of completion.