sscs-ose/sscs-chipathon-2025 #79: Analog/AI Track: Silicon Souls

Subhampal9, 2025-07-16:
Hello everyone,
We are team Silicon Souls. Our proposal for the Anlog/AI track is a design and layout of a Quadrature Output Two Stage Differential Ring VCO.
 

-  Repository -[https://github.com/ekta568/silicon_souls](url)

d-m-bailey, 2025-07-23:
@Subhampal9 the above link is invalid (display is correct, but points to this page. Maybe the repo was originally private?). Can you edit it with the correct link? https://github.com/ekta568/silicon_souls

Subhampal9, 2025-08-01:
@d-m-bailey I'm not sure why the link is not working, both the repos exist and are public repos

Subhampal9, 2025-08-13:
> [@Subhampal9](https://github.com/Subhampal9) Please update the first comment in this issue according to the instructions here [#101](https://github.com/sscs-ose/sscs-chipathon-2025/issues/101).
> 
> Be sure to include a progress tracker so that we can judge the likelihood of completion.