sscs-ose/sscs-chipathon-2025 #82: Digital Building Blocks - The leftovers

anotherlin, 2025-07-16:
Project is to design a full adder (a, b, and carry-in as inputs and sum and carry-out as outputs) standard cell, that can be used to implement simple ripple carry adders. In particular, carry-in pad would be at the rightmost of the standard cell and carry-out pad leftmost. So an adder would be a somewhat long row of full adder cells.

We can make this team ("the leftovers") for anyone who just wants to design a single standard cell, not necessarly a full adder.
Or I would gladly join another team that would accept my small project, thank you.

anotherlin, 2025-08-12:
Updated today with switch level Verilog simulation and basic schematics: https://github.com/anotherlin/sscs-chipathon-2025-D_19_the_leftovers/tree/main

d-m-bailey, 2025-08-13:
@anotherlin 
Please update the first comment in this issue according to the instructions here #101.

Be sure to include a progress tracker so that we can judge the likelihood of completion.

anotherlin, 2025-08-16:
@d-m-bailey I've updated the first comment according to the instructions. I don't have slides ready. Pin requirements are not applicable for standard cells?

d-m-bailey, 2025-08-17:
@anotherlin Look at some of the other digital teams to see what they've done in regards to pin lists.