sscs-ose/sscs-chipathon-2025 #90: MOSbius Track - Onchip Team

AlexMantilla1, 2025-07-23:
Hello there!

This is the issue to track our progress. We are Onchip Team!

**Team Members:**

- Alex Mantilla <alex-mantilla> **(M-I)**
- Sebastián Suárez **(B)**
- Jose Algarin **(B)**
- Ramón Sarmiento **(B)**
- Julián Romero **(B)**
- Fabián Chacón **(B)**
- Jorge Angarita **(M)**
- Ricardo Vergel **(M-I)**
- Luisa Dovale **(M-I)**
- And some occasional contributor*

**Team background**
Academic Experience:
**(M):** MSc or MSc Student at UIS
**(B):** BSc Student at UIS
Work Experience:
**(I):** Experience on industry

This project proposes the integration of a modular spiking neuron for implementation in MOSbius, targeting the 2025 Chipathon. The proposed architecture comprises a low-pass filter realized via a time-domain integrator based on a ring-oscillator VCO, in addition to a comparator and an output spike generation circuit. Each constituent block will be developed as a separate module within the MOSbius framework, enabling users to configure and evaluate them independently or integrate them into a complete spiking neuron for the construction of simple neural networks.

Also, you can see the project proposal video presentation at:
https://youtu.be/Y6UKVFCD-yw 

We are planning to report the updates in a github repo here:
https://github.com/Gior-gio/Chipathon-2025-OnChipTeam.git

peterkinget, 2025-07-24:
Thanks for your proposal. 

Please add a work distribution across the team members and a timeline. 

How will you test and demonstrate this block. 

There is another team proposing neuron circuits. Consider talking to them so you can collaborate on putting neurons together possibly on one test chip.

Gior-gio, 2025-07-25:
Hi, I'm another member of this group.

1) We just uploaded a new doc to keep track of our progress.
2) To test the final implementation, you should connect all the modules first. Some parts of it require external resistors to control the weight of the inputs. These voltages can be kept at a constant value for a basic test. But you should test it using a pulse train, which is more like a real application and could be made by an external wave generator.  (We're not sure if it's possible to use the MOSBius chip to generate it)

Gior-gio, 2025-08-08:
Hi, I'm Jorge from the Onchip team.

The neuron and its modules have been uploaded to the GitHub repository.

Additionally, you can watch an extended version of the schematic review in the following video:
https://youtu.be/cXW3zaoYFbs.

peterkinget, 2025-08-09:
@Gior-gio 

some feedback on your slides:

- please estimate the number of pins you need and the area

We will share how to share that information efficiently for our go/no-go review in the coming week.

d-m-bailey, 2025-08-13:
@AlexMantilla1 Thanks for the tag. The information looks well organized. 

Could you check the **progress** (my bad) tracker link? It appears to be a copy of the proposal video.
We will use the progress tracker to determine if you are on schedule to tapeout.

AlexMantilla1, 2025-08-13:
That's right! we're sorry about that. We have fixed this detail. Thanks for checking!

Gior-gio, 2025-08-25:
On behalf of our group, we would like to inform you that due to scheduling issues, we will not be continuing with our project at Chipathon 2025. Thank you very much for the opportunity.

d-m-bailey, 2025-11-24:
Your team was originally on the MOSbius track, right?

Working through LVS. Flattening layout while keeping the lower cell text causes text shorts. For example `to_gate` and `ASIG5V` of `io_secondary_5p0`

It looks like the layout contains dummy devices that aren't present in the schematic. Is it possible to add the dummy devices to the schematic so LVS will pass?

d-m-bailey, 2025-11-24:
New Axx_Onchip issue will be create so close this old one that was for the MOSbius track