sscs-ose/sscs-chipathon-2025 #98: [`run_GL` script] Move from Miniconda to Miniforge for update to TOC for Anaconda.org

amisapta15, 2025-07-30:
Due to recent update to [TOS](https://www.anaconda.com/legal/terms/terms-of-service) of anaconda, installation of miniconda fails with error

```
11.93 CondaToSNonInteractiveError: Terms of Service have not been accepted for the following channels. Please accept or remove them before proceeding:
11.93     • https://repo.anaconda.com/pkgs/main
11.93     • https://repo.anaconda.com/pkgs/r
11.93 
11.93 To accept a channel's Terms of Service, run the following and replace `CHANNEL` with the channel name/URL:
11.93     ‣ conda tos accept --override-channels --channel CHANNEL
```
See this [issue](https://github.com/pytorch/pytorch/pull/158370) for details. 

@MrHighVoltage Can we please update the `run_GL.sh` script 

https://github.com/iic-jku/IIC-OSIC-TOOLS/blob/ieee_chipathon/_build/images/iic-osic-tools/skel/dockerstartup/scripts/run_GL.sh#L25

with 
```
# Download and install Miniforge
    wget -O miniforge.sh "https://github.com/conda-forge/miniforge/releases/latest/download/Miniforge3-$(uname)-$(uname -m).sh"
    bash miniforge.sh -b -p "$MINICONDA_DIR"
```

d-m-bailey, 2025-07-30:
@hpretl Note that anaconda's terms of service have changed. (This was a big issue at one of the companies I consult for). You might want to get your school's lawyers to take a look at the TOS.

In March 2024, Anaconda updated its Terms of Service, introducing a commercial license requirement for organizations with 200 or more employees, regardless of whether they are for-profit, non-profit, or governmental entities. This change impacts the use of the entire Anaconda Distribution, including core Python tools and data science libraries, even when used indirectly through automation, containers, or cloud notebooks. Educational institutions are exempt for curricular activities (teaching, learning, research), but non-curricular research activities, especially if exceeding the 200-employee threshold, require a commercial license. 
Here's a more detailed breakdown:
Key Changes:
Commercial License Requirement:
Organizations with 200 or more employees or contractors need a paid license to use Anaconda's software for "Organizational Use". 
Definition of Organizational Use:
This includes any use of Anaconda's software by an organization, whether for development, research, production, or other purposes. 
Impact on Indirect Usage:
The change affects not just direct downloads and installations but also indirect use through containers, cloud notebooks, and other integrations. 
[Educational](https://www.google.com/search?cs=0&sca_esv=49fffdab3ad49e72&sxsrf=AE3TifOMkPBjKrnKOLUy3jg_DGHB1z73vw%3A1753902740732&q=Educational&sa=X&ved=2ahUKEwi64aLupOWOAxVYklYBHWSSES4QxccNegQIFBAB&mstk=AUtExfBawyA0IkB3VmI4R6bYa1f3Z8RxM1m5PDf5rUgzuFfDMyv3tI_yfDgXf8y3G51KXdHseLDEKC6r_8XNftw91JbgpUnd9VfXhWNFgO7EubYDyV0HS-rf3TeuzPCg4bfFvCtmmG2Q4DyGBe0_WIup5A7NegudMz5bwkUZqlBLVe1PVPM&csui=3) Exemptions:
Educational institutions are exempt for curriculum-based courses and related research, even if they have 200+ employees. 
Research Restrictions:
Non-curriculum research activities at educational institutions (especially if exceeding the employee threshold) may require a commercial license. 
What's Affected:
The entire [Anaconda Distribution](https://www.google.com/search?cs=0&sca_esv=49fffdab3ad49e72&sxsrf=AE3TifOMkPBjKrnKOLUy3jg_DGHB1z73vw%3A1753902740732&q=Anaconda+Distribution&sa=X&ved=2ahUKEwi64aLupOWOAxVYklYBHWSSES4QxccNegQIMBAB&mstk=AUtExfBawyA0IkB3VmI4R6bYa1f3Z8RxM1m5PDf5rUgzuFfDMyv3tI_yfDgXf8y3G51KXdHseLDEKC6r_8XNftw91JbgpUnd9VfXhWNFgO7EubYDyV0HS-rf3TeuzPCg4bfFvCtmmG2Q4DyGBe0_WIup5A7NegudMz5bwkUZqlBLVe1PVPM&csui=3), including:
Core Python tools
Data science libraries like [Pandas](https://www.google.com/search?cs=0&sca_esv=49fffdab3ad49e72&sxsrf=AE3TifOMkPBjKrnKOLUy3jg_DGHB1z73vw%3A1753902740732&q=Pandas&sa=X&ved=2ahUKEwi64aLupOWOAxVYklYBHWSSES4QxccNegQIOhAB&mstk=AUtExfBawyA0IkB3VmI4R6bYa1f3Z8RxM1m5PDf5rUgzuFfDMyv3tI_yfDgXf8y3G51KXdHseLDEKC6r_8XNftw91JbgpUnd9VfXhWNFgO7EubYDyV0HS-rf3TeuzPCg4bfFvCtmmG2Q4DyGBe0_WIup5A7NegudMz5bwkUZqlBLVe1PVPM&csui=3), [NumPy](https://www.google.com/search?cs=0&sca_esv=49fffdab3ad49e72&sxsrf=AE3TifOMkPBjKrnKOLUy3jg_DGHB1z73vw%3A1753902740732&q=NumPy&sa=X&ved=2ahUKEwi64aLupOWOAxVYklYBHWSSES4QxccNegQIOhAC&mstk=AUtExfBawyA0IkB3VmI4R6bYa1f3Z8RxM1m5PDf5rUgzuFfDMyv3tI_yfDgXf8y3G51KXdHseLDEKC6r_8XNftw91JbgpUnd9VfXhWNFgO7EubYDyV0HS-rf3TeuzPCg4bfFvCtmmG2Q4DyGBe0_WIup5A7NegudMz5bwkUZqlBLVe1PVPM&csui=3), [Scikit-learn](https://www.google.com/search?cs=0&sca_esv=49fffdab3ad49e72&sxsrf=AE3TifOMkPBjKrnKOLUy3jg_DGHB1z73vw%3A1753902740732&q=Scikit-learn&sa=X&ved=2ahUKEwi64aLupOWOAxVYklYBHWSSES4QxccNegQIOhAD&mstk=AUtExfBawyA0IkB3VmI4R6bYa1f3Z8RxM1m5PDf5rUgzuFfDMyv3tI_yfDgXf8y3G51KXdHseLDEKC6r_8XNftw91JbgpUnd9VfXhWNFgO7EubYDyV0HS-rf3TeuzPCg4bfFvCtmmG2Q4DyGBe0_WIup5A7NegudMz5bwkUZqlBLVe1PVPM&csui=3)
Conda environment management 
What's Not Affected:
Projects using open-source alternatives that don't rely on binaries directly distributed by Anaconda. 
Individuals using Anaconda for personal, non-commercial projects. 
Educational institutions using Anaconda for curricular activities, including teaching, learning, and research. 
Navigating Anaconda Licensing Changes: What You Need to ...
Sep 7, 2024 — So, what changed? The heart of the changes lies in Anaconda's definition of "Organizational Use." According to the new ...

DataCamp
Retrospective on Anaconda's 2024 Licensing Changes
May 11, 2025 — Starting in March 2024, Anaconda, Inc. updated its Terms of Service to introduce a new Commercial Edition License. Und...

Licenseware

Anaconda’s 2024 Licensing Changes and Alternatives
Feb 20, 2025 — The 2024 Licensing Changes Anaconda updated its Terms of Service, introducing a requirement for organizations with 200...

Python in Plain English

Show all

amisapta15, 2025-07-30:
@d-m-bailey Thanks for the message. As @d-m-bailey mentioned, we aren't distributing not using any anaconda-binaries. This update actually changes the mini-conda installation to mini-forge installation without accpeting the TOC, which seems to be people (issue linked above from PyTorch) are following. 

Here is the script in question. maybe, @MrHighVoltage  , we can remove `-c anaconda` from the conda install. Everything is installed from conda-forge as default in miniforge manager. Hope it is ok. 

```                                                    
#!/bin/bash
set -e  # Exit on error

# === Step 1: Basic environment setup ===
unset PYTHONPATH
unset LD_LIBRARY_PATH

# Define base directory for conda installation
BASE_DIR="$HOME/conda-env"
MINICONDA_DIR="$BASE_DIR/miniconda3"
ENV_NAME="GLdev"

export PATH="$MINICONDA_DIR/bin:$PATH"

# === Step 2: Check if environment is already set up ===
if [ -d "$MINICONDA_DIR/envs/$ENV_NAME" ]; then
    echo "Existing $ENV_NAME environment detected. Skipping setup."
else
    echo "$ENV_NAME environment not found. Starting setup..."

    # Create base directory
    mkdir -p "$BASE_DIR"
    cd "$BASE_DIR"

    # Download and install Miniconda
    # Download and install Miniforge
    wget -O miniforge.sh "https://github.com/conda-forge/miniforge/releases/latest/download/Miniforge3-$(uname)-$(uname -m).sh"
    bash miniforge.sh -b -p "$MINICONDA_DIR"

    # Source Conda
    source "$MINICONDA_DIR/etc/profile.d/conda.sh"

    # Create the environment
    conda create -y -n "$ENV_NAME" python=3.10

    # Activate the environment
    conda activate "$ENV_NAME"

    # Install packages
    conda install -y jupyter jupyterlab notebook nbclassic \
        jupyter_server_ydoc jupyter_server_fileid \
        numpy=1.24 matplotlib pip \
        -c litex-hub -c conda-forge -c anaconda

    # Register the kernel
    python -m ipykernel install --user --name="$ENV_NAME"

    # Pip packages
    pip install glayout
    pip install "klayout>=0.28,<0.29"
    pip install svgutils

    echo "Setup complete!"
fi

# === Step 3: Launch Jupyter ===
echo "Launching Jupyter Lab..."
source "$MINICONDA_DIR/etc/profile.d/conda.sh"
conda activate "$ENV_NAME"
jupyter lab --ip=0.0.0.0 --no-browser --port=8888
```

MrHighVoltage, 2025-07-31:
The change will be submitted to our repo, but we will not yet trigger a rebuild, since it is working for now.

amisapta15, 2025-08-13:
This can be closed now right?

MrHighVoltage, 2025-08-26:
Yes, this is done!

closed 2025-08-26 by hpretl