sscs-ose/sscs-chipathon-2025 #123: CAD: Klayout pfet pcell for GF180

peterkinget, 2025-08-15:
@atrokmabrains 

When you try to instantiate a pfet pcell in klayout for GF180, it does not appear. The work around is to instantiate an nfet and then change its properties to a pfet. 

The fossi-chat discussion is here: https://matrix.to/#/!DnBsMaeQdLuqxWYMZv:fossi-chat.org/$LEosCyQEiU88b-EICZPtY0L-EtBAaionyzBfy1LWJto?via=fossi-chat.org&via=matrix.org&via=gitter.im

Royce mentioned this might have to do with the default size of the pmos.

peterkinget, 2025-08-15:
No errors on the terminal but this info is there:

```
PDK: gf180mcuD /foss/designs > klayout
(22, 0)
## gf180mcu PDK Pcells loaded.
['/foss/pdks/ciel/gf180mcu/versions/f2e289da6753f26157a308c492cf990fdcd4932d/gf180mcuD/libs.tech/klayout/tech/pymacros', '/foss/tools/klayout/pymod', '/usr/lib/python312.zip', '/usr/lib/python3.12', '/usr/lib/python3.12/lib-dynload', '/usr/local/lib/python3.12/dist-packages', '/usr/lib/python3/dist-packages', '/headless/.klayout/python']
## gf180mcu PDK Pcells loaded.
['/foss/pdks/gf180mcuD/libs.tech/klayout/tech/pymacros', '/foss/pdks/ciel/gf180mcu/versions/f2e289da6753f26157a308c492cf990fdcd4932d/gf180mcuD/libs.tech/klayout/tech/pymacros', '/foss/tools/klayout/pymod', '/usr/lib/python312.zip', '/usr/lib/python3.12', '/usr/lib/python3.12/lib-dynload', '/usr/local/lib/python3.12/dist-packages', '/usr/lib/python3/dist-packages', '/headless/.klayout/python']
```

peterkinget, 2025-08-15:
So I tried to recreate the problem, but it disappeared. But it has to do with the initial values in the pcell parameters for the pfet. The default width is too small I believe. If I change it to 1um then the cell appears. And now it keeps working even after restarting klayout ...

atorkmabrains, 2025-08-16:
That's Great. It seems it was some form of misconfiguration.

atorkmabrains, 2025-08-16:
We could close this issue then.

d-m-bailey, 2025-08-16:
@peterkinget I think klayout stores a lot of settings in `~/.klayout/klayoutrc`. Do you see a default `pfet` width stored there?